tianyoujian/MZDPI #4: I2C Disabled

somacoder, 2019-05-17:
The use of this screen (Version-A 2.8 inch Touch Screen) appears to disable any use of I2C for other devices. Is there any way to disable use of I2C and still maintain functionality of the screen?

Thanks!

Andreeved, 2019-05-22:
sry for my russian english..i'm using mzdpi-vga-zero 2.8inch on zero w and have same problem. but solved)
add "dtoverlay=i2c-gpio,i2c_gpio_sda=25,i2c_gpio_scl=26" in boot/config.txt 
gpio25-sda-22 pin
gpio26-scl- 37 pin
bus i2c-3

closed 2019-05-22 by somacoder

RyanWoirol, 2020-10-24:
> sry for my russian english..i'm using mzdpi-vga-zero 2.8inch on zero w and have same problem. but solved)
> add "dtoverlay=i2c-gpio,i2c_gpio_sda=25,i2c_gpio_scl=26" in boot/config.txt
> gpio25-sda-22 pin
> gpio26-scl- 37 pin
> bus i2c-3

I'm sorry but I'm not following this.  Is this 4 separate lines?  The spacing is weird and I'm not familiar with linux at all.

toygar-12, 2021-01-20:
Is this same for Version-C?