tianyoujian/MZDPI #6: Yellow tint on latest raspbian.

suraj022, 2019-07-19:
The screen works but has yellow tint on the latest raspbian buster. works fine on older raspbian stretch build.

paulloft, 2019-07-30:
in your installation script you need to add 0 after x to all addresses to get this:

```
DTBFILE=/boot/bcm2708-rpi-0.dtb
if [ -f "$DTBFILE" ];then
	dtc -W no-unit_address_vs_reg -I dtb -O dts -o /boot/tmp.dts $DTBFILE

	if grep -q "brcm,pins = < 0x08 0x07 >;" /boot/tmp.dts; then
		if grep -q "brcm,function = < 0x01 >;" /boot/tmp.dts; then
	      		echo "\nZERO";
			sed /boot/tmp.dts -i -e "s/brcm,function = < 0x01 >;/brcm,function = < 0x06 >;/"
			cp /boot/bcm2708-rpi-0.dtb /boot/bcm2708-rpi-0.bak	
			dtc -W no-unit_address_vs_reg -I dts -O dtb -o $DTBFILE /boot/tmp.dts 
		fi
	fi
	rm /boot/tmp.dts
fi
```
Also in the new versions of Rasbian added new config files for Pi Zero and ZeroW, which are not described in the installation script. If you use them, then you will also have to add them to the installation script by analogy with the rest.
new config files
```
bcm2708-rpi-zero.dtb
bcm2708-rpi-zero-w.dtb
```

suraj022, 2019-07-30:
Thanks for the reply.
I've tried what you suggested. but unfortunately the problem still persists. I've added 0 after x to all addresses and also added scripts for new config files but the screen is still yellowish.

paulloft, 2019-07-30:
Try to start the uninstall first (in the uninstall script you need to do the same steps) and then the installation again.
 I literally ran into the same problem today, and adding 0 helped me. 
The thing is that if you manually run the command `dtc -W no-unit_address_vs_reg -I dtb -O dts -o /boot/tmp.dts /boot/<you_board>.dtb` and search for the phrase `brcm, pins =` in `/boot/tmp.dts` then you will notice that the addressing has changed and added 0 after x.
Сheck it, it is possible that your problem is not related to it at all

suraj022, 2019-08-01:
Tried with a fresh install of raspbian buster. yellow tint still exist. maybe it has some other incompatibilities.

will36, 2019-08-04:
@paulloft, thank you for your troubleshooting but I have also tried this with no luck. Still whites are yellow and blacks are blue. I have latest Raspbian Buster - 2019-07-10.
@suraj022 Did you have any luck fixing the yellow?

suraj022, 2019-08-05:
@will36 unfortunately not. But I'am using retropie which uses raspbian stretch and it works fine in that build.

jtme, 2019-08-23:
same issue with raspbian Strech, if it is up to date (4.19.66). Fresh install - colors all ok, after apt-get upgrade - yellow tint.

paulloft, 2019-08-23:
try manually rebuilding your .dtb config, run
```
dtc -W no-unit_address_vs_reg -I dtb -O dts -o /boot/tmp.dts /boot/<your_board>.dtb
```
open /boot/tmp.dts in nano or vim editor and find lines similar to

brcm,pins = < 0x08 0x07 >
brcm,function = < 0x01 >

and replace

brcm,pins = < 0x08 0x07 >
brcm,function = < 0x0**6** >

I noticed that on different versions of Linux these lines may differ by the presence or absence of zeros and spaces, so the installer script does not work. Look for something similar in your file.

Now you can try to pack everything back in .dtb, run
```
dtc -W no-unit_address_vs_reg -I dts -O dtb -o  /boot/<your_board>.dtb /boot/tmp.dts
```
I also recommend checking your /boot/config.txt for lines at the end of the file
```
dtparam=spi=on
dtoverlay=ads7846,penirq=27,swapxy=1,xmin=200,xmax=3850,ymin=200,ymax=3850
display_rotate=3
dtoverlay=mzdpi
framebuffer_width=640
framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x07f003
hdmi_timings=480 0 41 20 60 640 0 5 10 10 0 0 0 60 0 32000000 1
```

jtme, 2019-08-24:
no success. Tried both. Rebuilding .dtb file on stretch and also re-installing fresh with buster.  Both cases failed for me. But as it works with stretch without updates i am certain it is not hardware issue. 

I can just hope that original script will be updated to keep up with latest versions of raspbian.

paulloft, 2019-08-25:
I just tried a clean installation of Raspian Stretch on my raspberry zero w with apt-update, and everything works for me. 
I have a display version of B, maybe this has something to do with the version?

I also rewrote the installer for myself, it is based on overlays, and there is no need to change system dts files anymore even after updating. Try it, maybe it will help https://github.com/paulloft/raspberry/blob/master/display/mzdpi-vga/mzdpi-vga-autoinstall-online

jtme, 2019-08-25:
thanks @paulloft. I was meanwhile following #9 description and as well made a clone in my own repository, and with that updated install script everything works ok with Raspian Buster and RPI Zero W.

texas-home-pros, 2020-04-20:
@jtme I cloned your repo and attempted to install and still cannot get the screen to work; would i have to solder it down to get it to work or can it just slip through the holes for testing?

jtme, 2020-04-20:
@texas-home-pros - definitely needs to be soldered on.

texas-home-pros, 2020-04-20:
Is there a wrong way to solder it on or can you just put it directly
through the holes from the screen to the back of the Pi Zero and solder?

jtme, 2020-04-20:
Please follow the manufacturer's suggestion. I personally chose to skip the header due to space for the project
[https://imgur.com/a/MU0x1Xf](url)

cacry, 2020-06-26:
Also
    sudo raspi-gpio set 8 a2
    sudo raspi-gpio set 7 a2
fixes the issue

ekx, 2020-09-18:
If you want the yellow/blue tint to go away as early as possible in the startup process create a file called "fixscreen.sh" somewhere (e.g. in the home directory) containing the following
`#!/bin/bash
sudo raspi-gpio set 8 a2
sudo raspi-gpio set 7 a2`
Make the script executable
Then create a file called "fixscreen.service" in /lib/systemd/system containing the following
`[Unit]
Description=Fix Screen
Before=basic.target
After=local-fs.target sysinit.target
DefaultDependencies=no

[Service]
Type=oneshot
ExecStart=/home/pi/fixscreen.sh

[Install]
WantedBy=basic.target`

aaronkogan, 2021-02-08:
Pi Zero Rev 1.3 the same yellow tint. I found a bug script mzdpi-vga-autoinstall-online
in DTBFILE=/boot/bcm2708-rpi-zero.dtb section missed space:
if grep -q "brcm,pins = < 0x08 0x07>;" /boot/tmp.dts; then => if grep -q "brcm,pins = < 0x08 0x07 >;" /boot/tmp.dts; then

dts was rebuilded with warnings and not helped:

/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/mmc@7e202000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/i2s@7e203000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/i2s@7e203000:dmas: cell 2 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/spi@7e204000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/spi@7e204000:dmas: cell 2 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/hdmi@7e902000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/mmc@7e300000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/mmcnr@7e300000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (dmas_property): /soc/smi@7e600000:dmas: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (mboxes_property): /soc/firmware:mboxes: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (phys_property): /soc/usb@7e980000:phys: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/dsi@7e209000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/dsi@7e700000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/vec@7e806000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/usb@7e980000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/hdmi@7e902000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/v3d@7ec00000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/csi@7e800000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (power_domains_property): /soc/csi@7e801000:power-domains: cell 0 is not a phandle reference
/boot/bcm2708-rpi-zero.dtb: Warning (thermal_sensors_property): /thermal-zones/cpu-thermal:thermal-sensors: cell 0 is not a phandle reference