tianyoujian/MZDPI #21: FATAL ERROR: Unrecognized check name "gpios_property"

Razzorback225, 2021-02-22:
Hi, these errors appears when trying to use the auto-install-online script.
`FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"
grep: /boot/tmp.dts: No such file or directory
rm: cannot remove '/boot/tmp.dts': No such file or directory
FATAL ERROR: Unrecognized check name "gpios_property"`

The Display is not working except for the backlight.

I tried to install the drivers with the latest version of raspbian lite.

KERNEL VERSION :
Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

TheLastMillennial, 2021-03-02:
Good news, I believe I've found a solution to the errors!
**Short answer:** Manually update [device tree compiler](https://launchpad.net/ubuntu/+source/device-tree-compiler/1.4.7-1/+build/15279269/+files/device-tree-compiler_1.4.7-1_armhf.deb).

**Long explanation:**
I was getting the same error on my Banana Pi M2 Zero running Raspian stretch. However, it worked on my Raspberry Pi Zero W (v1.3). I checked the [source code](https://github.com/tianyoujian/MZDPI/blob/master/vga/mzdpi-vga-autoinstall-online#L96) where the errors were appearing and I notice that dtc (device tree compiler) seemed to be the culprit. I check the dtc version on both the Raspberry pi and the Banana pi with `dtc -v`

This returned version 1.4.7 on the Raspberry Pi but 1.4.4 on the Banana Pi. Unfortunately, running `sudo apt-get install device-tree-compiler` just returned a message that dtc was already the latest version. I found where all the dtc versions are stored and downloaded [device-tree-compiler_1.4.7-1_armhf.deb](https://launchpad.net/ubuntu/+source/device-tree-compiler/1.4.7-1/+build/15279269/+files/device-tree-compiler_1.4.7-1_armhf.deb).

First I ran `sudo ./mzdpi-vga-uninstall` (found under MZDPI/vga) just in case the failed MZDPI install messed something up. Then I navigated to where I downloaded the .deb file and ran `sudo apt install ./device-tree-compiler_1.4.7-1_armhf.deb`. No errors occurred during the update and when I tried `dtc -v` again, it returned 1.4.7!

Finally, I ran `sudo ./mzdpi-vga-autoinstall-online` and it installed with no errors! Unfortunately, I don't have a display myself to test if it actually works (I wanted to check if the drivers were compatible in the first place before I bought), but I don't see any reason why it shouldn't work now.

Versions:
- Banana Pi M2 Zero Kernel: Linux 4.4.55-BPI-M2Z-Kernel armv7l
- Raspberry Pi Zero (v1.3) Kernel: Linux 5.4.83+ armv6l

Razzorback225, 2021-03-02:
@TheLastMillennial Wow thanks for your research. I'll test this out this morning to verify it's working 😃