tianyoujian/MZDPI #24: Error install MAZDPI pi zero

clrmsc, 2021-12-07:
`pi@raspberrypi:~/MZDPI/vga $ sudo ./mzdpi-vga-autoinstall-online
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xserver-xorg-input-evdev is already the newest version (1:2.10.6-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xinput-calibrator is already the newest version (0.7.5+git20140201-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
matchbox-keyboard is already the newest version (0.2+git20160713-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/virtgpio: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/i2c0mux: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpu: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/axiperf: simple-bus unit address format error, expected "7e009800"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/firmware: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/power: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/gpiomem: simple-bus unit address format error, expected "7e200000"
/boot/tmp.dts: Warning (simple_bus_reg): /soc/fb: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/vcsm: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/sound: missing or empty reg/ranges property
/boot/tmp.dts: Warning (simple_bus_reg): /soc/virtgpio: missing or empty reg/ranges property`

clrmsc, 2021-12-07:
How fix it pls help!

0hweav, 2022-01-16:
you must have Buster OS installed I believe, I installed Volumio on a Pizero and cannot get the drivers installed....but on Buster OS I can go through the steps and the install completes...maybe missing base packages or something? I'm still new